HKUST-Aerial-Robotics/DenseSurfelMapping #4: rviz always crashes

DYYYYYYYYY, 2019-05-16:
Hello, rviz always crashes when I load rviz_config.rviz in the last step. Why is this?

DYYYYYYYYY, 2019-05-16:
When I loaded rviz_config.rviz, the following error occurred after rviz crashed.

terminate called after throwing an instance of 'Ogre::InternalErrorException'
  what():  OGRE EXCEPTION(7:InternalErrorException): Cannot create GL vertex buffer in GLHardwareVertexBuffer::GLHardwareVertexBuffer at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/OgreGLHardwareVertexBuffer.cpp (line 47)

WANG-KX, 2019-05-16:
Hello,
How about you manually add the topic from dense_surfel_mapping? And what's your ubuntu, ROS, and RVIZ version?

DYYYYYYYYY, 2019-05-16:
I open rviz via rosrun rqt_rviz rqt_rviz and load rviz_config.rviz with the open config option.
My ROS version is
 * /rosdistro: kinetic
  * /rosversion: 1.12.14
My rviz version is
[ INFO] [1557975521.968744314]: rviz version 1.12.17
[ INFO] [1557975521.968783295]: compiled against Qt version 5.5.1
[ INFO] [1557975521.968793213]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1557975522.102393911]: Stereo is NOT SUPPORTED
[ INFO] [1557975522.102473766]: OpenGl version: 3 (GLSL 1.3).

WANG-KX, 2019-05-16:
Try
```rviz rviz_config.rviz``` ?

DYYYYYYYYY, 2019-05-16:
Do I manually add a topic after I open it? In addition, my computer does not have a GPU. Does this matter?

WANG-KX, 2019-05-16:
Have you already run the code and then launch the rviz? Or just launch the rviz and it crashes?

DYYYYYYYYY, 2019-05-16:
The picture after I manually added the topic is as follows, and it is different from the dense picture in the demo.
[image]

DYYYYYYYYY, 2019-05-16:
I followed the order you wrote, and finally opened rviz.

closed 2019-05-16 by WANG-KX

luzhaozhang2276, 2019-09-14:
@WANG-KX  @DYYYYYYYYY 
你好,请问我在最后一步运行`rviz rviz_config.rviz`命令后,rviz里的Fixed Frame出现了错误,rviz画面里也没有地图的更新信息,请问你知道怎么解决吗?
[image]

DYYYYYYYYY, 2019-10-20:
@luzhaozhang2276 尝试把Global Options 里的Fixed Frame 改成 world试试看