HKUST-Aerial-Robotics/DenseSurfelMapping #5: I can't open the saved PLY file with CloudCompare. What is the reason?

hululuhu, 2019-05-27:


hululuhu, 2019-05-28:
The reason is that the PLY file is incompletely saved. Press crtl + c to exit the process and save the PLY file. Because the PLY file is large, the PLY file will be saved incompletely.

WANG-KX, 2019-05-28:
Are you pressing ctrl+c for many times before the saving finishes? Actually, I don't know any other methods to "save the map before quitting". Maybe adding a ROS SERVICE to save is a good option.

closed 2019-07-08 by WANG-KX

WANG-KX, 2019-07-13:
Just notice in my code, that it actually support saving PLY file when you publish a rostopic called "save_map" with string msg. Check the code here: https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/c9e32e2b14b3028e8f7491524bf924dc9bd32534/surfel_fusion/src/ros_node.cpp#L26