HKUST-Aerial-Robotics/DenseSurfelMapping #8: vins-supported map save?

MoonUniverse, 2019-07-17:
Hi,thank you for your work! However,the code of vins-supported seem can't save the map.What do I need to do to save the map?

Thank you!

WANG-KX, 2019-07-17:
It can save the map. The node also subscribes a rostopic called "save_map". Just publish a ros message in string with the path you want to save.
Follow the code to know more: https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/468fdc79d64d94b07ebfd027bccf8f339898a031/surfel_fusion/src/ros_node.cpp#L26.

One simple example: After all the fusion is done, type the following in your terminal:
rostopic pub /save_map std_msgs/String "/home/user/model.ply"

MoonUniverse, 2019-07-17:
Thank you very much for your reply! 
Map saved successfully!

closed 2019-07-17 by MoonUniverse

Quitino, 2019-12-02:
Hi@MoonUniverse @WANG-KX  ,thank you for your work!  @WANG-KX , Does vins-supported version of the code support online browsing of Mesh map?

Thank you!