HKUST-Aerial-Robotics/DenseSurfelMapping #9: 我不想让图片随镜头动

wangyuanbiubiubiu, 2019-07-29:
您好,我运行这个程序的时候,弹出来的小窗里会显示出surfel,但是我不想让他跟随我的镜头,我该怎么将他固定在原点呢?就是我想看全貌图

WANG-KX, 2019-07-29:
comment the [line](https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/f3b8ddab9f0c5342f8fd8dc0a30ed48a3cffed4f/surfel_fusion/src/fusion_functions.cpp#L1001) and the [line](https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/f3b8ddab9f0c5342f8fd8dc0a30ed48a3cffed4f/surfel_fusion/src/fusion_functions.cpp#L1005).

wangyuanbiubiubiu, 2019-07-29:
ok 谢谢!

wangyuanbiubiubiu, 2019-07-29:
我注释掉了以后确实发现没有了哪个小窗,但是我想看全貌图的话,只能等全部建图完了才能看么?我现在的surfel是会自己不断更新,后面的面片就没有了

WANG-KX, 2019-07-29:
using RVIZ?

wangyuanbiubiubiu, 2019-07-29:
对,我用的rviz
效果图 
[image]
不知道使用哪个话题看?
[image]

WANG-KX, 2019-07-29:
the depth is terrible...
I suggest you fix the depth quality before fusion.
To visualize the fused surfel in point cloud, try surfel_fusion/pointcloud

wangyuanbiubiubiu, 2019-07-29:
好的,我去看看

wangyuanbiubiubiu, 2019-07-30:
I now use their own depth map, but I found that the program can't deal with the depth of my figure. I checked the, may be stored in my depth map numerical problem. For example, I estimate the depth of the pixel is 20 m and I was in the pixel position assignment in the MAT, could you tell me the correct depth map should be inside the assignment?

[image]

wangyuanbiubiubiu, 2019-07-30:
我用中文说一下,就是这是我自己生成的深度图,深度的估计是没有问题的,但是我输入到这个程序里发现并没有输出结果.我现在如果估计出这个像素的深度是x m,我就将一个CV_32FC1 矩阵中对应地方赋值x,我猜想是不是这个程序要求的深度图不是这种定义方式?

WANG-KX, 2019-07-30:
Depth map is defined as CV_32FC1.
Please check your launch file, including camera parameters, and fusion max distence. Also, why you think your depth map is ok?

wangyuanbiubiubiu, 2019-07-30:
Ok, I'll check these things. Thanks!

wangyuanbiubiubiu, 2019-07-30:
I found some useful parameters. Now I'm a car driving on the ground. Should I adjust the parameters here? I feel that the threshold is too small. What parameters should I adjust?
[image]

My current situation is that my surfel always follows my camera and then disappears.

wangyuanbiubiubiu, 2019-07-30:
[image]
[image]

This is the debug diagram I output at runtime. I wonder if I need to adjust some parameters because my scene is too big.

wangyuanbiubiubiu, 2019-07-30:
and what means drift_free_poses?

WANG-KX, 2019-08-19:
Well... Sorry for the late response. Just found this issue. If you have a new issue, please open a new one.
The map fusion quality largely depends on the quality of input depth and the associated camera poses. In your debug image, maybe your depth is too noisy to extract any useful surfels. Can you please show me the depth of this frame and also the corresponding intensity image?
Drift－free-pose is used to describe the quality of your odometry system. You can find details in the paper, basically, more accurate odometry system has larger drift-free-poses.

closed 2020-02-18 by WANG-KX