HKUST-Aerial-Robotics/DenseSurfelMapping #10: 文章下载

ShaoQingGong, 2019-07-31:
你好，我想问一下，Real-time Scalable Dense Surfel Mapping这篇文章在哪儿能下载到？

WANG-KX, 2019-07-31:
Hi, I have uploaded the paper in the readme of the master branch.

closed 2019-07-31 by WANG-KX

ShaoQingGong, 2019-08-05:
你好！
能直接把文章发到我邮箱么？大陆这边很难翻墙到dropbox下载东西。
谢谢！

Closed #10.

ShaoQingGong, 2019-08-09:
凯旋，你好！
       我现在想在VINS-Mono上跑DenseSurfelMapping，但是看github上只有在ORB-SLAM2上跑DenseSurfelMapping的指令，并没有在VINS-Mono上跑DenseSurfelMapping的指令。因此，想问一下，在VINS-Mono上跑DenseSurfelMapping的具体指令或流程，急用，麻烦你了！
                                                                                                                                                                                                                  2019年8月9日
                                                                                                                                                                                                                        龚少庆

WANG-KX, 2019-08-10:
少庆你好！ git上有vins supported branch，请切换到那个branch。 凯旋 在2019年08月09日 11:35，ShaoQingGong 写道： 凯旋，你好！ 我现在想在VINS-Mono上跑DenseSurfelMapping，但是看github上只有在ORB-SLAM2上跑DenseSurfelMapping的指令，并没有在VINS-Mono上跑DenseSurfelMapping的指令。因此，想问一下，在VINS-Mono上跑DenseSurfelMapping的具体指令或流程，急用，麻烦你了！ 2019年8月9日 龚少庆

ShaoQingGong, 2019-08-19:
凯旋，你好！
       由于各种原因，我这边没法在VINS-Mono上跑DenseSurfelMapping，现在准备在ORB-SLAM2上跑DenseSurfelMapping，环境和各种依赖包都配好了，关于数据集我们使用的是您提供的数据集，但是关于要输入深度图的数据集我们没有，而用PSMNet来生成深度图比较麻烦，而我们想快速地看一下效果。您能把关于深度图的数据集信息发给我么？我们程序现在唯一缺的就是此语句中的“depth_path = path_name + "/depth_0/%06d.npy" % image_index”npy文件，希望能得到您的帮助，谢谢！
                                                                                                                                                                                                                               2019年8月19日
                                                                                                                                                                                                                                        龚少庆

在 2019-08-09 11:35:46，"龚少庆" <18810296853@163.com> 写道：

        凯旋，你好！
       我现在想在VINS-Mono上跑DenseSurfelMapping，但是看github上只有在ORB-SLAM2上跑DenseSurfelMapping的指令，并没有在VINS-Mono上跑DenseSurfelMapping的指令。因此，想问一下，在VINS-Mono上跑DenseSurfelMapping的具体指令或流程，急用，麻烦你了！
                                                                                                                                                                                                                  2019年8月9日
                                                                                                                                                                                                                        龚少庆

WANG-KX, 2019-08-19:
你好，
我记得ORB-SLAM2版本可以跑kitti的数据集，这是一个开源的数据集。depth 文件我已经删除了，由于比较大，还是建议你自己生成一下。
附录是生成的代码，需要你根据自己的环境改一下path什么的。
祝好，
凯旋