HKUST-Aerial-Robotics/DenseSurfelMapping #12: Some issue about sync topics

YiChenCityU, 2019-09-03:
Hi, when I try this code on my own device (d435i), I met this problem. The depth msg cannot be received by surfel_fusion. I think the sync policy has some bug. Thanks very much. 
[image]

WANG-KX, 2019-09-08:
It is hard to find if there are any bugs given only one image.
Could you provide more information?

YiChenCityU, 2019-09-10:
I found that I my data always has "image_time >pose_reference_time", which cause endless loop.

WANG-KX, 2019-09-10:
well... You are using the ORB branch?
Also, the intrinsic parameter is very confusing. Are you sure the fx and fy are different so much? And cy is 237.3. What's the resolution of your depth camera?

closed 2019-11-04 by YiChenCityU

qmoji, 2019-11-15:
相机留在最后启动