HKUST-Aerial-Robotics/DenseSurfelMapping #14: Enable Colorized pointcloud

jessychen1016, 2019-10-12:
Really appreciate your work on the Surfel-Fusion. It is not really an issue but is there a possible way that i can actually turn the plain/boring single colored pointcloud into something nicer like endowing the true color of the object to that point?

WANG-KX, 2019-10-20:
Hi, we also received similar suggestions from our coworker. It will be implemented in the future.
The reason it does not support is that in our previous applications, we simply use grayscale images.

jessychen1016, 2019-10-25:
Hey, thank you for your feedback! Looking forward to your future work!

Shame-fight, 2020-07-30:
Do I need to modify some code, or run `roslaunch surfel_fusion vins_realsense.launch`  directly after running vins-fusion? I can’t receive superpixel norm and spimage after running with the camera, so I can’t create a map. I hope it doesn’t bother you. Thanks for help!
@WANG-KX @jessychen1016