HKUST-Aerial-Robotics/DenseSurfelMapping #20: Have trouble in rviz the real-time map

XanxusCrypto, 2020-01-14:
Thanks for sharing the mapping tool!
While I ran the Kitti00 dataset, it seemed like that the surfel-mapping thread worked normally, and I could observe colorful surfels in the debug-show window as shown in Fig 1. But when I turned on the Rviz launch file, some troubles was met. 
Firstly the rivz file can not be read properly as I ran "rosrun rviz rviz path-to-rviz-file" in Fig 2 and 3, and  the pointclouds were all colorful whether active or inactive (which should be gray as I watched other issues or the youtube video uploaded), finally the view point did not follow the camera motion. As I was not familar with the rviz options, would you plz share the right way to launch rviz?

[image]
[image]
[image]

XanxusCrypto, 2020-01-17:
> you can open the rviz without any config file and then load the config file in the rviz window.

@WANG-KX  感谢您的回复！鉴于折腾了几天代码比较头疼这里直接用中文与你交流吧。
您给的KITTI00数据集我已经跑通了，上面提出的问题是由于rviz文件没法读入，所以只能手动录入了，感谢指导。然后这几天我想测试一下代码运行EuRoC数据集的效果，由于电脑没有GPU，所以没有用您分享的深度学习框架计算disparity image，用的是Ros里Dense_Stereo_Node的SGBM算法来得到双目disparity image，然后导入到建图算法中，大体框架如下：

Dense_Stereo_Node：Rosbag play EuRoC V101.bag，输出双目图像和disparity image
ORB_SLAM2: 接受双目图像，输出path，pose和回环时间戳
Surfel_Mapping: 接受左目和disparity image，path，pose和回环时间戳，运行程序建图。

现在ORB_SLAM2接收V101.bag的图像时首先会提示
cannot find the reference keyframe! time difference > 1000000000... 可能是因为rosbag play时间戳的问题，报错一次之后后面ORB_SLAM2程序是正常运行的，所以稍微调试后也没继续管。

Surfel_Mapping我是在rosbag play前打开的，但是输出后会各种报错，一些常见的log如图一所示。

通过输出KITTI00数据集学习出的深度值和Dense_Stereo_Node匹配出的视差转成深度，可见数量级相差很远。KITTI00 depth image像素点深度值在几千到几万。
而双目匹配恢复出的深度值量级很小，基本相差了四五个数量级，我乘以100000后显示如图二所示，已经转成了cv::MAT 32FC1格式。
所以我想首先解决深度图的问题。通过MVDepthNet训练出的disparity image需要rescale吗？我看代码，计算出local_surfel后再与near_distance和far_distance比较（代码给出是0.5m和30m，rviz来看kitti街道宽度也差不多是10m尺度），所以双目匹配恢复出的深度如何scale呢？还望指教。
如果github上交流不方面的话，也可以邮件交流，我的邮箱是vs09970997@zju.edu.cn。

[image]
[image]

qq1962572025, 2020-02-13:
我也遇到了不能读入rviz文件的问题，可以告诉我你是怎么手动导入的吗？谢谢。

XanxusCrypto, 2020-02-18:
> 我也遇到了不能读入rviz文件的问题，可以告诉我你是怎么手动导入的吗？谢谢。

@qq1962572025  新打开一个terminal，输入rviz打开rviz，然后将global options里的Fixed Frame改成world，就可以正常接收显示发布的点云和路径信息了。

closed 2020-06-05 by WANG-KX