HKUST-Aerial-Robotics/DenseSurfelMapping #23: Uneven results for Reconstruction

qq1962572025, 2020-02-17:
The reconstruction rsult is quiet uneven.As shown after,a small part of  the provided dataset is much better than others.
[image]
I'm curious why.

WANG-KX, 2020-02-18:
It is very abnormal. 
From the image, I guess you are running KITTI. Seems the trajectory is right but the fusion is wrong. Have you visualized your depth map before fusion? And what's your method for depth map estimation.

qq1962572025, 2020-02-18:
> It is very abnormal.
> From the image, I guess you are running KITTI. Seems the trajectory is right but the fusion is wrong. Have you visualized your depth map before fusion? And what's your method for depth map estimation.

I have found the reason.This is because I adopted the wrong image transfer method. I changed the transfer method in surfer_fusion to the ORB-SLAM2 method.

closed 2020-02-18 by qq1962572025