HKUST-Aerial-Robotics/DenseSurfelMapping #24: Having trouble running TUM dataset

qq1962572025, 2020-02-18:
I want to run program on an indoor dataset to observe the result of the reconstruction. I modify the camera Intrinsic in the launch file for node surfer_fusion, and rewrite the node that sends the picture like this:
	    depth = cv::imread(PATH+s1, -1);
	    left = cv::imread(PATH+s2,-1);
	    cvtColor( left, left, CV_BGR2GRAY ); 
	    depth.convertTo(depth,CV_32F);
            sensor_msgs::ImagePtr msg_1 = cv_bridge::CvImage(std_msgs::Header(), "mono8", left).toImageMsg();
	    sensor_msgs::ImagePtr msg_2 = cv_bridge::CvImage(std_msgs::Header(), "32FC1", depth).toImageMsg();
ORB-SLAM2 runs well as usual,but surfer_fusion cannot create any surfel,is there other part of the program that should be modified?
[image]

qq1962572025, 2020-02-25:
The unit of the depth map in the code is meter, so TUM data needs to be divided by factor.

closed 2020-02-25 by qq1962572025

allenthreee, 2021-12-11:
> 

can you give more details about the factor about TUM dataset, thank you

canglangzhige, 2022-05-14:
> The unit of the depth map in the code is meter, so TUM data needs to be divided by factor.

您好，我也在尝试运行TUM的数据集。我没有写数据的发布节点，而是直接使用tum 的rosbag直接发布数据，我的运行结果如下：
[image]
[image]
但是orb的点云在rviz中显示不出来，显示为：
[image]
运行surfel_fusion节点时，rviz没有出现任何surfel，我的几个终端显示：
[image]
按您说的，我在surfel_map.cpp文件中把深度图的单位进行了转换：
image.convertTo(image,CV_32F,mDepthMapFactor);
mDepthMapFactor 应该是取0.0002吧（5000我也试过），我看TUM的5000值对应距离1米，但还是不行，请问您有什么建议吗？