HKUST-Aerial-Robotics/DenseSurfelMapping #30: 利用PSMNET生成kitti_odometry_00的深度图像，最大深度是128。然后利用vins-fusion进行该算法的测试，结果非常不好，不知道什么原因？

yueshukun, 2020-05-19:
具体测试截图如下：
[image]
[image]
生成的深度图如下：
[image]
求高手解答，谢谢！

WANG-KX, 2020-05-19:
The result fro PSMNet is a disparity map. Convert to depth maps (meters) before any usage.

closed 2020-05-31 by WANG-KX

708949503, 2020-06-29:
您好，请问您之前提出的那个问题解决了吗？
OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows)

zheng-xiao, 2022-05-24:
请问，您的视差图还保存着吗？我使用PMSNet生成的视差图精度不好