HKUST-Aerial-Robotics/DenseSurfelMapping #33: terminate called after throwing an instance of 'std::bad_alloc'

dbdxnuliba, 2020-07-07:
when I running  
roslaunch surfel_fusion vins_realsense.launch

and mapping 3d map
after a time while, 
the terminal shows :+1: 

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[surfel_fusion-1] process has died [pid 24384, exit code -6, cmd /home/dell/denseSurfel_vino_supptorted_src/devel/lib/surfel_fusion/surfel_fusion ~image:=/camera/color/image_raw ~depth:=/cam

dbdxnuliba, 2020-07-07:
[image]
this is the caputer

WANG-KX, 2020-07-17:
Sorry, cannot help you debug this error without more information.

zeroAska, 2022-09-22:
I have run into the same error. Have you figured it out?