HKUST-Aerial-Robotics/DenseSurfelMapping #34: NO superpixel norm

PumpkinRun, 2020-07-28:
Thank you for your work，and release it. When i use ORB_slam2   model RGBd  
to run Kitti dataset, sequence 00, there is no problem. And i can get the ideal result. But i processing our outdoor road data, it  don't extract the super pixels, and  don't generate the point cloud. I don't know which problem it is. Can you help me to solve it?　I am looking forward your reply!!!   Thank you!

[image]

[image]

[image]

[image]

Shame-fight, 2020-07-30:
I have the same problem,and the spimage also can not received when using mynteye camera,but the surfel_color02.bag is right.so whether need to modify some code to solve these problems, except for vins_realsense.launch.looking forward for your reply.thanks~ @dvorak0 @WayneTimer @groundmelon @shaojie

PumpkinRun, 2020-07-30:
Hi~~,At present, I can extract super pixels. When reading depth map, pay attention to  the unit of depth image  is meter. After changing the information of reading depth map in ' publisher.py ' , I can get super pixels, but the reconstruction effect is very poor, running orbslam2 rgbd mode , Core dump was appeared. Because the quliaty  of depth image is poor?  Hope to provide some ideas and solutions. Thanks!!!  I am looking forward to your replay!

[image]
[image]

Shame-fight, 2020-07-31:
Successfully running the program is good news, although the effect is not very good. Can you tell me where I need to modify pubilsher.py? I am using a stereo camera, not an rgb-d camera. I cannot run this program normally at the moment, Thanks in advance.
@PumpkinRun

Shame-fight, 2020-07-31:
Successfully running the program is good news, although the effect is not very good. Can you tell me where I need to modify pubilsher.py? Are there other codes that need to be modified? I am using a stereo camera, not an rgb-d camera. I cannot run this program normally at the moment, Thanks in advance. Do you have e-mail or wechat or qq,It is more convenient to communicate with chat tools.
@PumpkinRun

WANG-KX, 2020-08-01:
Hi, I am here for help. 
@PumpkinRun can you convert the depth maps into point clouds and show me? From the normal map, it seems quite noisy.

WANG-KX, 2020-08-01:
@Shame-fight The messages should be well synchronized. If you do not receive any messages please go to https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/77f65b1cfd3390ca50f1c94b1f0f250aeec2a00b/surfel_fusion/src/surfel_map.cpp#L83, https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/77f65b1cfd3390ca50f1c94b1f0f250aeec2a00b/surfel_fusion/src/surfel_map.cpp#L103, https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/77f65b1cfd3390ca50f1c94b1f0f250aeec2a00b/surfel_fusion/src/surfel_map.cpp#L93, and https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/77f65b1cfd3390ca50f1c94b1f0f250aeec2a00b/surfel_fusion/src/surfel_map.cpp#L205  for debugging.
I do not know much about your senesors and configuration, so please refer to the example bag and configure your settings.

PumpkinRun, 2020-08-05:
> Hi, I am here for help.
> @PumpkinRun can you convert the depth maps into point clouds and show me? From the normal map, it seems quite noisy.
Hello, Thank you for your reply. I had converted the depth maps into point clouds and discoveried it is terrible.

1. This is kitti 00 sequence  image00 , 000000.png and 000000.npy.

[image]

[image]

2. This is owner data including image_raw and point clouds.

[image]
[image]

So, It should be the problem of depth map, which makes the reconstruction result poor

WANG-KX, 2020-08-05:
Hi, @PumpkinRun  not sure what's on the ground. It does not seems right, right?