HKUST-Aerial-Robotics/DenseSurfelMapping #37: the problem  about using my own camera for reconstruction

zhhz9706, 2020-10-20:
Thank you for your excellent work. When I use DenseSurfelMapping, I use VINS-Fusion, and the camera uses Realsense D435i, but when I use it, DenseSurfelMapping does not work. There is no information when I open spImage in rviz. Do I need to synchronize the posture information with the image and depth map in the DenseSurfelMapping code before fusing it? If so, how should it be synchronized?
Thanks again for your work!

WANG-KX, 2020-10-20:
Hi,
take a look at [this function](https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/a22aba45e86d629cfccdbb15152cc0608598880b/surfel_fusion/src/surfel_map.cpp#L185). This function synchronizes all necessary messages.

zhhz9706, 2020-10-20:
> 嗨，
> 看看[这个功能](https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/a22aba45e86d629cfccdbb15152cc0608598880b/surfel_fusion/src/surfel_map.cpp#L185)。此功能同步所有必要的消息。

Hi, thank you for your reply. In the `void SurfelMap::synchronize_msgs() function`, you have performed the time stamp synchronization work, but I see that you have defined the `typedef message_filters::sync_policies::ExactTime<sensor_msgs::PointCloud, nav_msgs::Path, nav_msgs earlier ::Odometry> exact_policy;` But it is not used. I tried to synchronize the rgb image and the depth map in D435i, but it still cannot be used. What is the reason for this? Do I need to use `messag_filters` for time synchronization before calling the callback function?
Looking forward to your reply!

zhhz9706, 2020-12-08:
> Hi,
> take a look at [this function](https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping/blob/a22aba45e86d629cfccdbb15152cc0608598880b/surfel_fusion/src/surfel_map.cpp#L185). This function synchronizes all necessary messages.

Hi，
Can I use the odometry provided by Realsense T265 and the Depth provided by D435i as the input of DenseSurfelMapping?

greymaner, 2021-04-06:
@zhhz9706 I also want to use both t265 and d435i to generate surfel maps, can you give me some advice? thx!