HKUST-Aerial-Robotics/DenseSurfelMapping #40: Using SGM

RamadanAhmed, 2021-01-07:
Hi,
Thanks for the project, I want to ask if I can use sgm from https://github.com/fixstars/libSGM to calculate disparity maps or there is certain methods that should be used

WANG-KX, 2021-01-11:
hi, sorry for the late reply.
Any depth estimation method can be used for map fusion. The better the depth, the better the map you can get.
Regarding SGM, I'm not sure how good this "libsgm" is. If the disparity you get is discrete, I don't think the quality can be good.

RamadanAhmed, 2021-01-11:
Thanks for answering,
I didn't try it yet, but I was asking if there are any constraints/methods that should be applied before using other depth estimation techniques

closed 2021-01-11 by RamadanAhmed